gazebosim/docs #14: Consolidate Library READMEs with installation tutorials

chapulina, 2020-04-29:
Some libraries have repeated content in their installation tutorials and READMEs
Choose one or the other for each library to have

ahcorde, 2020-08-13:
Consolidated install documentation for the following libraries (Moved install instructions to the tutorials folder):

 - [ ] ign-fuel-tools https://github.com/ignitionrobotics/ign-fuel-tools/pull/101
 - [ ] ign-gui https://github.com/ignitionrobotics/ign-gui/pull/106
 - [ ] ign-math https://github.com/ignitionrobotics/ign-math/pull/147
 - [ ] ign-physics https://github.com/ignitionrobotics/ign-physics/pull/95
 - [ ] ign-plugin https://github.com/ignitionrobotics/ign-plugin/pull/27
 - [ ] ign-rendering https://github.com/ignitionrobotics/ign-rendering/pull/125
 - [ ] ign-transport https://github.com/ignitionrobotics/ign-transport/pull/160

chapulina, 2020-08-13:
Thanks for taking this on, @ahcorde !

I commented on one of the PRs, but for completeness I'll mention it here too that we should be linking to the published link of those installation tutorials, such as

https://ignitionrobotics.org/api/math/6.4/install.html

The reasons are:

* The `README`s are rendered not only on GitHub but also (once #58 is fixed) on pages like this, and relative links to the tutorials won't work: https://ignitionrobotics.org/libs/math
* The tutorials may not always render correctly on GitHub, their canonical location is on the `ignitionrobotics.org` site

chapulina, 2020-12-23:
Reopening as we're still missing some libraries:

- [ ] `ign-cmake` (note that its tutorials aren't being deployed yet, see https://github.com/ignitionrobotics/docs/issues/53)
- [ ] `ign-common`
- [ ] `ign-tools` (this one is tricky, because it doesn't use `ign-cmake`, so https://github.com/ignitionrobotics/docs/issues/53 is also tricky)
- [ ] `ign-gazebo`
- [ ] `ign-msgs`
- [ ] `ign-launch`

mabelzhang, 2020-12-24:
> The tutorials may not always render correctly on GitHub, their canonical location is on the `ignitionrobotics.org` site

I'm consolidating `ign-common` at the moment.

Do we want to keep the relative section links, e.g. the links to different types of installs under the Overview section here:

They don't jump correctly in the canonical location
https://ignitionrobotics.org/api/math/6.8/install.html#binary-install

But they jump correctly on GitHub
https://github.com/ignitionrobotics/ign-math/blob/ign-math6/tutorials/installation.md

chapulina, 2020-12-24:
> Do we want to keep the relative section links

Yeah I recently noticed that too. I think we should remove them since they don't work on the main site.