gazebosim/docs #28: releases: Page not rendering?

EricCousineau-TRI, 2020-05-30:
I tried viewing the following page:
<https://ignitionrobotics.org/docs/releases>

However, this is what I see:
> [image]

Javascript error from Developer console:
```
ERROR TypeError: "a.docsInfo.pages[a.version] is undefined"
```

Browser: Firefox 76.0.1
OS: Ubuntu 18.04

chapulina, 2020-06-01:
Yeah, the correct URL is https://ignitionrobotics.org/docs/all/releases

I remember there were links missing the `/all/` but I can't find them now. Do you remember how you got to this link? The link on the [Ignition Documentation](https://ignitionrobotics.org/docs) page looks correct.

EricCousineau-TRI, 2020-06-01:
Same as here: https://github.com/ignitionrobotics/docs/issues/29#issuecomment-636953759

I currently don't remember how I got to it two days ago :(

chapulina, 2020-06-01:
I think I found one place where these links can be coming from: other pages. Like the Roadmap's link to the Releases page is wrong:

https://ignitionrobotics.org/docs/all/roadmap

I think it just needs to be updated here:

https://github.com/ignitionrobotics/docs/blame/master/roadmap.md#L49

There are probably others in the same situation.

EricCousineau-TRI, 2020-06-02:
Ah, maybe that was it - thanks!!!

PR #74 (2020-08-28): Fix link to releases page
Resolves #28
closed 2020-08-28 by chapulina