gazebosim/docs #29: "Edit" button redirects to wrong link?

EricCousineau-TRI, 2020-05-30:
As I was investigating #28, I tried clicking on the Edit Icon ([image]), and it took me to the following link:
<https://github.com/ignitionrobotics/docs/blob/master/>

I think the correct link should have been either:
<https://github.com/ignitionrobotics/docs/blob/master/releases.md>
or
<https://github.com/ignitionrobotics/docs/tree/master/> ?

Current commit: 198e1f9

chapulina, 2020-06-01:
I think that the page you're trying to edit should show a 404 instead of a title and pencil:

https://ignitionrobotics.org/docs/releases

This is the correct page and its edit button redirects to the correct GitHub file for me:

https://ignitionrobotics.org/docs/all/releases

:arrow_down: 

https://github.com/ignitionrobotics/docs/blob/master/releases.md

EricCousineau-TRI, 2020-06-01:
Huh... now that you mention it, I have no idea how I came across the wrong link <https://ignitionrobotics.org/docs/releases> - I don't think I typed it in, but in Firefox History, I can't seem to identify what my sites were referrers were at the time of the visit...

So yup, that sounds good! Want me to re-purpose this issue, or close it and open a more accurate one?

chapulina, 2020-06-01:
> Want me to re-purpose this issue, or close it and open a more accurate one?

I'm fine either way, thanks!

EricCousineau-TRI, 2020-06-02:
Repurposed!

ahcorde, 2020-08-18:
@nkoenig I fixed this on the backend. Do you mind to have a look?

scpeters, 2020-08-24:
Fix has been merged but not deployed

chapulina, 2020-08-27:
Fixed: https://ignitionrobotics.org/docs/banana

chapulina, 2020-08-27:
Spoke too soon, this pattern is still broken: https://ignitionrobotics.org/docs/latest/banana

CC @ahcorde

ahcorde, 2020-08-28:
This PR fixed

 - /docs/<wrong_release>/install
 - /docs/citadel/< wrong tutorial >

https://gitlab.com/ignitionrobotics/web/web/-/merge_requests/20

chapulina, 2020-08-31:
Thanks, I think we're all set now.

closed 2020-08-31 by chapulina