gazebosim/docs #32: libs: Include SDFormat?

EricCousineau-TRI, 2020-06-02:
For this page: https://ignitionrobotics.org/libs
Is it possible (or does it make sense) to include SDFormat too? (or something like "Parsing", "Modeling", "Modeling Formats", ...?)

FTR This seems to be more comprehensive (but lower level?):
https://github.com/ignitionrobotics/docs/blob/198e1f957195cb38945f352dc136076052f7db82/contributing.md#repository-list

(Stemming off of this post: https://community.gazebosim.org/t/gsod-2020-ignition-robotics-on-boarding-guide/576/2?u=ericcousineau-tri )

chapulina, 2020-06-02:
Yeah these are good questions. Given that SDF is older than Ignition, the approach we've been taking has been to treat it as not a sibling but an older cousin. So, still part of the family :slightly_smiling_face:

Sticking with that approach, I guess it would make sense to keep all SDFormat documentation in https://sdformat.org, and link to that from the libs page. That may need some custom code on the website's server.

ahcorde, 2020-08-18:
@nkoenig I created a PR in the backend. Do you mind to have a look?

scpeters, 2020-08-24:
Fix has been merged but not deployed

chapulina, 2020-11-16:
Current status:

* SDF listed on these pages:
    * https://ignitionrobotics.org/docs
    * https://ignitionrobotics.org/libs
* README / Changelog rendered on this entry page: https://ignitionrobotics.org/libs/sdformat
* Links to source code correctly point to https://github.com/osrf/sdformat
* API doc links are currently broken, for example: https://ignitionrobotics.org/api/sdformat/8.0/index.html
    - [ ] These links should point to libSDFormat's own API docs, e.g. http://osrf-distributions.s3.amazonaws.com/sdformat/api/8.2.0/index.html

chapulina, 2021-03-31:
We could also add SDFormat to the API doc [upload script](https://github.com/ignitionrobotics/docs/blob/master/tools/build_docs.sh), which is used in the nightly doc uploads.

Here's SDFormat's own [upload docs script](https://github.com/osrf/sdformat/blob/144f82cf8b994abd2a566a7bbe6ae873a9d92403/cmake/upload_doc.sh.in).