gazebosim/docs #34: Install from source for Focal

chapulina, 2020-06-11:
The [install from source tutorial for Bionic](https://ignitionrobotics.org/docs/citadel/install_ubuntu_src) doesn't fully work on Focal. For example, the `libdart6.*-dev` packages need to be renamed to `libdart.*-dev`, without the `6`.

So we need to either update the tutorial to apply for both versions, or create a new tutorial for Focal.

Any thoughts on how to arrange this info, @maryaB-osr ?

j-rivero, 2020-06-11:
Maybe I can do something to help with the instructions. One option would be to generate a `libdart-dev` in our own releases (at leat Bionic) that will do nothing but pull `libdart6-dev` to the system.

We could even deprecate `libdart6-dev` and make it a transitional package to `libdart-dev` which mimic the official Ubuntu package in Focal.

A couple of details more: dart upstream has its own PPA in Ubuntu and uses versioned `-dev` package names (I can try contacting them to align the whole naming). There are many `-dev` packages that need to be changes or deprecated.

maryaB-osr, 2020-06-11:
A quick fix could be just following the step:
```
The command below will install all dependencies in Ubuntu Bionic:
```
With another alternative step:
```
The command below will install all dependencies in Ubuntu Focal:
```

chapulina, 2020-06-11:
Thank you both! @j-rivero , maybe we can punt on making those deb changes, because I imagine we may find out more differences between both distros.

@maryaB-osr , that sounds reasonable to me. The tutorial mentions Bionic all over the place though, so I think we'll need to change that.

chapulina, 2020-06-11:
I just remembered that for Focal we should be able to skip [Install compiler requirements](https://ignitionrobotics.org/docs/citadel/install_ubuntu_src#install-compiler-requirements).

chapulina, 2020-06-15:
@maryaB-osr , I just assigned it to you, but let me know if you think you won't get to it anytime soon. Thanks!

maryaB-osr, 2020-06-19:
@chapulina Sorry, just saw this, yes I can do it. Maybe not today but otherwise ASAP

PR #69 (2020-08-26): Citadel source install for Focal
Resolves #34.
closed 2020-08-26 by chapulina