gazebosim/docs #53: Missing libs on docs page

chapulina, 2020-08-13:
This page:

https://ignitionrobotics.org/docs

Is missing ign-tools and ign-cmake.

In a related note, it would be nice to alphabetize them so they're easier to find. In the future we could consider having an icon for each one so they're easier to spot.

ahcorde, 2020-08-24:
- [x] Alphabetize libraries: https://gitlab.com/ignitionrobotics/web/web/-/merge_requests/14

chapulina, 2020-12-22:
Generating tutorials for `ign-cmake`: https://github.com/ignitionrobotics/ign-cmake/pull/137

mabelzhang, 2020-12-24:
Are we excluding `sdformat` from this list? Its tutorial link on https://ignitionrobotics.org/libs is broken.

chapulina, 2020-12-28:
> Are we excluding sdformat from this list? 

That's being addressed as part of https://github.com/ignitionrobotics/docs/issues/32.

azeey, 2024-07-18:
Fixed by #441

closed 2024-07-18 by azeey