gazebosim/docs #55: Single index with tutorials from all libraries

chapulina, 2020-08-13:
Our tutorials are currently scattered around:

* Generic top-level ones are here: https://ignitionrobotics.org/docs
* And each version of each library has its own tutorials page, i.e. https://ignitionrobotics.org/api/gazebo/3.2/tutorials.html

This is tough to navigate, especially for new users who don't know where to begin from. So we should make them easier to discover.

One idea is to offer a view with links to all the tutorials for all the libraries. That would be similar to how we have a browse page for all Gazebo-classic tutorials from all categories:

http://gazebosim.org/tutorials/browse

ahcorde, 2020-08-24:
I added a new section on the web-site with all the tutorials availables in the `index.yaml` for each Ignition version:

[image]

azeey, 2025-02-25:
We could potentially used https://www.breathe-doc.org/ for this.