gazebosim/docs #110: Create social preview image banners for all libraries

chapulina, 2020-10-17:
GitHub supports adding a custom open graph images to each repository so that they display nicely when shared on social media:

https://github.blog/2019-04-17-custom-open-graph-images-for-repositories/

It would be nice to create specific images for each library. I suggest adding the image to each repository with the name `banner.png`, and one of the maintainers can make it the official banner.

It would be cool to have banners specific to each library that highlight their functionality.

Here's GitHub's template:

[image]

Dasux, 2025-01-13:
why is this issue still open

azeey, 2025-01-13:
I don't think any of the libraries have a banner yet. Are you interested in working on it?