gazebosim/docs #113: d

traversaro, 2020-11-01:


traversaro, 2020-11-01:
cc @wolfv @Tobias-Fisher @j-rivero

chapulina, 2020-11-03:
> Binaries compiled against ignition-lib version x.y.z can run without problems with a compile ignition-lib with the same major version and that has a version >= x.y.z .

That's correct.

> if a binary compiled against ignition-lib 1.11.2 can run with a binary 1.11.0, and I would say that it is not possible as in release 1.11.2 new symbols can be introduced w.r.t. release 1.11 .

You're right, the ABI compatibility is only when going from a lower version to a higher version, but not the other way around.

Note though that we (strive to) bump the minor when adding new API. So in your specific example, where only the patch version was bumped, ABI compatibility can be assumed in both directions.

---

Thank you for bringing this up. We mention the use of SemVer in a couple places:

* https://ignitionrobotics.org/docs/all/releases
* https://ignitionrobotics.org/docs/all/roadmap#feature-roadmap

But strictly speaking, SemVer is only about API, not ABI. So I think we can state our ABI policy explicitly in one of those 2 places and make the corner cases you mentioned clear as well.

j-rivero, 2020-11-03:
> But it would be nice to a have a clear reference, to clarify corner cases (see for example [conda-forge/libignition-fuel-tools-feedstock#12 (comment)](https://github.com/conda-forge/libignition-fuel-tools-feedstock/pull/12#issuecomment-720069113), where we argue if a binary compiled against ignition-lib 1.11.2 can run with a binary 1.11.0, and I would say that it is not possible as in release 1.11.2 new symbols can be introduced w.r.t. release 1.11 .

World if full of interesting corner cases I've never though before. I agree with @chapulina comments here.

chapulina, 2021-05-11:
I tried to concisely document this on https://github.com/ignitionrobotics/docs/pull/183, I hope it's clear enough.

traversaro, 2021-05-12:
Thanks, I think that docs solve the issue!

PR #183 (2021-06-29): Versioning documentation
Closes #113

## Summary

Added more semver documentation, as well as tick-tock explanation (ported from http://gazebosim.org/#status).

**Note to maintainers**: Remember to use **Squash-Merge**

🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸🔸
closed 2021-06-29 by chapulina