gazebosim/docs #116: Document macOS caveats on binary installation tutorial

chapulina, 2020-11-05:
The install tutorial for macOS from source lists caveats and extra tips to get Ignition to work:

https://ignitionrobotics.org/docs/citadel/install_osx_src

While the binary install tutorial suggests that everything will just work out of the box:

https://ignitionrobotics.org/docs/citadel/install_osx

We should make sure that users installing from binaries are aware of caveats and any extra steps needed.

ahcorde, 2021-03-01:
- https://github.com/ignitionrobotics/docs/pull/150
 - https://github.com/ignitionrobotics/docs/pull/151

chapulina, 2021-03-01:
Thanks, @ahcorde . Are you working on more PRs or do these cover everything?

CC @scpeters

ahcorde, 2021-03-02:
These are the two issues that I have found. Maybe I can add the `gazebo_scene_viewer` as an alternative to visualize what's going on in the scene.

What do you think ?

chapulina, 2021-03-02:
I think the main problem is the sentence "_All libraries should be ready to use and the ign gazebo app ready to be executed._", which gives the impression that everything should just work. I think we should document in those tutorials that the 3D scene doesn't work and that `ign gazebo` should only be run headless, see https://github.com/ignitionrobotics/ign-gazebo/pull/477/ as a reference.

We should document this for Citadel, Dome and Edifice.

chapulina, 2021-11-15:
I see this is already documented

closed 2021-11-15 by chapulina