gazebosim/docs #148: Add a tutorial explaining how to contribute documentation

chapulina, 2021-02-17:
## Desired behavior

We want users to contribute documentation, but it's not always straightforward how to do it. We have some tips on generating API docs in the  [code contributing guide](https://ignitionrobotics.org/docs/all/contributing), but I think there's enough to explain about documentation to prompt its own tutorial.

## Alternatives considered

We can also keep this documentation together with the code contribution docs, but I think we could at least give it more context and make it easier to find.

## Implementation suggestion

Things to document:

* How the API docs and per-library tutorials are generated using doxygen and can be tested locally (some explanation already in each install tutorial, i.e. [here](https://ignitionrobotics.org/api/common/3.10/install.html))
* The fact that the API docs are [deployed nightly](https://github.com/ignitionrobotics/docs/blob/master/.github/workflows/nightly-upload.yml)
* How the docs on this site are using custom code on a private repository to be generated, are deployed on demand, and can't be tested locally yet (#85)

## Additional context

This came up in https://github.com/ignitionrobotics/docs/issues/145#issuecomment-780026297 and https://github.com/ignitionrobotics/docs/pull/147

peci1, 2025-03-26:
I've hit this again, 3 years later, not remembering a thing :-D

I didn't know how to build per-library docs. Then I somewhere saw the `make docs` target, but I figured I'd have to install all `main` branch dependencies to be able to CMake it. Uh-oh, no go... So I ended up commenting out most of the CMakeLists.txt so that only the docs generating part remained. This allowed me to do local docs testing, but the UX is quite bad (even worse now that people are encouraged to send PRs against `main`, which nobody has installed on his PC except maybe the core devs).

peci1, 2025-03-26:
What I was missing a lot was a high-level description saying:

> Gazebo libraries use Doxygen with Markdown. They use `\` instead of `@` for commands. This is how you make a link... This is how you reference another tutorial... This is how you include an image. No, you cannot use Mermaid for UML. Beware that -- becomes em dash outside code blocks, so use \\-- instead. You can/cannot use github-style admonitions. Beware that `code` in headings doesn't work. And so on...

Also, a link to https://www.doxygen.nl/manual/markdown.html would help a lot.

Next, the first place I visited to figure out how to contribute docs were the CONTRIBUTING.md files in gz-sim and docs. None of them told me a word about how to contribute docs. I must admit I haven't figured out that I'd learn that in the installation tutorial (because I don't want to install Gazebo, I almost certainly already have it).