gazebosim/docs #202: Erros are printed in terminal after launching buildind_robot.sdf

matosinho, 2021-09-25:
## Environment
* OS Version: Ubuntu 20.04
* Source or binary build?
  * Source
  * main (Fortress)
  * commit: 653b0fb58abd6ea8b3e3937e43d5aee509e125fc
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: default.
      - [ ] Sensor rendering error.
      - [x] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [x] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
      0a:00.0 VGA compatible controller [0300]: NVIDIA Corporation TU106 [GeForce RTX 2060 SUPER] [10de:1f06] (rev a1)
      direct rendering: Yes
      OpenGL vendor string: NVIDIA Corporation
      OpenGL renderer string: GeForce RTX 2060 SUPER/PCIe/SSE2
      OpenGL core profile version string: 4.6.0 NVIDIA 460.91.03
      OpenGL core profile shading language version string: 4.60 NVIDIA
      OpenGL core profile context flags: (none)
      OpenGL core profile profile mask: core profile
      OpenGL version string: 4.6.0 NVIDIA 460.91.03
      OpenGL shading language version string: 4.60 NVIDIA
      OpenGL context flags: (none)
      OpenGL profile mask: (none)
      OpenGL ES profile version string: OpenGL ES 3.2 NVIDIA 460.91.03
      OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

<details>  

```
19:29:01: Creating resource group General
19:29:01: Creating resource group Internal
19:29:01: Creating resource group Autodetect
19:29:01: SceneManagerFactory for type 'DefaultSceneManager' registered.
19:29:01: Registering ResourceManager for type Material
19:29:01: Registering ResourceManager for type Mesh
19:29:01: Registering ResourceManager for type Skeleton
19:29:01: MovableObjectFactory for type 'ParticleSystem' registered.
19:29:01: ArchiveFactory for archive type FileSystem registered.
19:29:01: ArchiveFactory for archive type Zip registered.
19:29:01: ArchiveFactory for archive type EmbeddedZip registered.
19:29:01: DDS codec registering
19:29:01: FreeImage version: 3.18.0
19:29:01: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
19:29:01: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,psb,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
19:29:01: PVRTC codec registering
19:29:01: ETC codec registering
19:29:01: Registering ResourceManager for type HighLevelGpuProgram
19:29:01: Registering ResourceManager for type Compositor
19:29:01: MovableObjectFactory for type 'Entity' registered.
19:29:01: MovableObjectFactory for type 'Light' registered.
19:29:01: MovableObjectFactory for type 'BillboardSet' registered.
[... 233 lines cut ...]
```

</details>

## Description
* Expected behavior: 
  * `building_robot.sdf` should work fine without printing errors.
* Actual behavior: 
    * `building_robot.sdf` works, but it is printing erros as soon as it opens, seems something related to sdformat 1.8.

## Steps to reproduce

1. clone this repo and run `cd docs`
2. run `ign gazebo fortress/tutorials/building_robot/building_robot.sdf` 

## Output

`building_robot.sdf` has these warning upon loading:
```bash
Error:   Could not find the 'robot' element in the xml file
         at line 109 in /build/urdfdom-VnCcob/urdfdom-1.0.4+ds/urdf_parser/src/model.cpp
Error [parser_urdf.cc:3255] Unable to call parseURDF on robot model
Error [parser.cc:820] parse as old deprecated model file failed.
[GUI] [Err] [Model.hh:73] Unable to unserialize sdf::Model
```

scpeters, 2021-09-27:
>   * The `left_wheel` has some unwanted text in `moving_robot.sdf` and `sensor_tutorial.sdf`, here is how the `ixx` is defined: `                  [Build your own robot](SDF_tutorial_link) tutorial. You can download the robot fro   <ixx>0.145833</ixx>`

I have a fix for this specific error in #204

scpeters, 2021-09-28:
I can't reproduce the errors

scpeters, 2021-09-28:
I will address the non-unique name warnings in https://github.com/ignitionrobotics/ign-gazebo/issues/1056

matosinho, 2021-09-28:
I'll update everything and test again to see if the errors are fixed them. The warnings appeared for you, but the errors don't?