gazebosim/docs #222: ROS 2 beginner tutorials using Ignition Gazebo

chapulina, 2021-12-20:
We should create introductory tutorials that will go into http://docs.ros.org/en/rolling/Tutorials.html and live in https://github.com/ros2/ros2_documentation. Tracking in this repo so we can more easily track with other Ignition tasks.

## Desired behavior

New ROS 2 developers can easily launch and tweak a full robot development environment even if they don't have a physical robot.

## Alternatives considered

We already have tutorials on https://ignitionrobotics.org/docs/fortress/ros2_integration, but that's not easy to find for ROS 2 beginners, and don't show a complete application.

## Implementation suggestion

Do some simple tasks with a turtlebot in an interesting world.

chapulina, 2022-01-28:
It would be nice to use the TurtleBot 3 for the tutorials. See this issue upstream:

* https://github.com/ROBOTIS-GIT/turtlebot3_simulations/issues/179

chapulina, 2022-03-01:
Adding Ignition support to more robots, so we can document it later:

* https://github.com/frankaemika/franka_ros2/pull/1/
* https://github.com/ROBOTIS-GIT/turtlebot3_simulations/pull/180

ahcorde, 2022-03-01:
Another one:
 - https://github.com/iRobotEducation/create3_sim/pull/148

ahcorde, 2022-03-01:
- Turtlebot 4: https://github.com/turtlebot/turtlebot4_simulator

chapulina, 2022-03-22:
Done in https://github.com/ros2/ros2_documentation/pull/2356

closed 2022-03-22 by chapulina