gazebosim/docs #225: Update the style guide to outline how namespaces should be used

adlarkin, 2022-01-04:
## Desired behavior
We currently do not have any rules/guidelines on how namespaces should be used. This could result in naming collisions between various projects. We should outline dos and don'ts of namespace usage for ignition projects.

## Alternatives considered
N/A (new feature)

## Implementation suggestion
Add a section to https://ignitionrobotics.org/docs/all/contributing#style-guides that outlines how namespaces should be included/used. Some guidelines were discussed in this thread: https://github.com/ignitionrobotics/ign-sensors/pull/177#discussion_r774794118

## Additional context
The original conversation for this was started in https://github.com/ignitionrobotics/ign-sensors/pull/177#discussion_r774794118

mjcarroll, 2022-02-01:
I think it would be helpful to generally follow the `cppcore` guidelines, specifically SF 6 & 7

https://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#sf6-use-using-namespace-directives-for-transition-for-foundation-libraries-such-as-std-or-within-a-local-scope-only

https://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#sf7-dont-write-using-namespace-at-global-scope-in-a-header-file

scpeters, 2022-02-01:
> I think it would be helpful to generally follow the `cppcore` guidelines, specifically SF 6 & 7 [...]

that makes sense, both avoiding `using namespace` statements at global scope in a header file and preferring local scope when possible

I don't know if there are any linters that can detect this

mjcarroll, 2022-02-07:
`clang-tidy` can get it: https://clang.llvm.org/extra/clang-tidy/checks/google-build-using-namespace.html