gazebosim/docs #230: I was trying to spawn models from fuel as mentioned in docs- https://ignitionrobotics.org/docs/citadel/fuel_insert

harshmahesheka, 2022-02-11:
## Environment
* OS Version: 
* Source or binary build?
  
  
* If this is a GUI or sensor rendering bug, describe your GPU and rendering system. Otherwise delete this section. 
    - Rendering plugin: [ogre | ogre2]. 
      - [ ] Sensor rendering error. 
      - [ ] GUI rendering error. 
    - EGL headless mode:
      - [ ] Running in EGL headless mode 
    - Generally, mention all circumstances that might affect rendering capabilities: 
      - [ ] running on a dual GPU machine (integrated GPU + discrete GPU)
      - [ ] running on a multi-GPU machine (it has multiple discrete GPUs)
      - [ ] running on real hardware
      - [ ] running in virtual machine
      - [ ] running in Docker/Singularity
      - [ ] running remotely (e.g. via SSH)
      - [ ] running in a cloud
      - [ ] using VirtualGL, XVFB, Xdummy, XVNC or other indirect rendering utilities
      - [ ] GPU is concurrently used for other tasks
        - [ ] desktop acceleration
        - [ ] video decoding (i.e. a playing Youtube video)
        - [ ] video encoding
        - [ ] CUDA/ROCm computations (Tensorflow, Torch, Caffe running)
        - [ ] multiple simulators running at the same time
      - [ ] other...
    - Rendering system info:
      - On Linux, provide the outputs of the following commands:
          ```bash
          LANG=C lspci -nn | grep VGA  # might require installing pciutils
          echo "$DISPLAY"
          LANG=C glxinfo -B | grep -i '\(direct rendering\|opengl\|profile\)'  # might require installing mesa-utils package
          ps aux | grep Xorg
          sudo env LANG=C X -version  # if you don't have root access, try to tell the version of Xorg e.g. via package manager
          ```
      - On Windows, run `dxdiag` and report the GPU-related information.
      - On Mac OS, open a terminal and type `system_profiler SPDisplaysDataType`. Copy the output here.
        
    - [ ] Please, attach the ogre.log or ogre2.log file from  `~/.ignition/rendering` 
    
<details>
  
```
# paste log here
```

</details>

## Description
* Expected behavior: 
* Actual behavior: 

## Steps to reproduce

1.
2.
3.

## Output

nkoenig, 2022-02-11:
Yes, that is on purpose. It was removed because it stopped working with some version of Ignition. More importantly, Fuel is designed to be used by more than just Ignition, and the button was out of place given that design criteria.

You can spawn models into simulation using the Resource Spawner plugin, or you can copy the SDF snippet from the Fuel website using the button that looks like `<>`.

scpeters, 2022-02-11:
if the button was removed on purpose, then we need to update the out-of-date tutorials

harshmahesheka, 2022-02-11:
I would love to contribute; could you please assign this to me.

scpeters, 2022-02-11:
> I would love to contribute; could you please assign this to me.

thank you!

chapulina, 2022-03-01:
> It was removed because it stopped working with some version of Ignition. 

I wasn't able to reproduce this issue with any version of Ignition, as demonstrated here: https://gitlab.com/ignitionrobotics/web/app/-/merge_requests/146#note_857910759

> More importantly, Fuel is designed to be used by more than just Ignition, and the button was out of place given that design criteria.

I don't think there was anything in the implementation of the button that was specific to Gazebo. Any application could add drag-drop support if they wanted.

> You can spawn models into simulation using the Resource Spawner plugin

The Resource Spawner is less feature-rich than the Ignition Fuel website when it comes to searching for and inspecting models. I think its only advantage is that it also shows local models. 

> you can copy the SDF snippet from the Fuel website using the button that looks like <>.

That's not convenient when you already have a running simulation.

closed 2022-06-06 by chapulina