gazebosim/docs #245: No debug setting for colcon

onurtore, 2022-04-18:
During the source installation process there is no separate configuration for how to set the debug mode.

I think the solution is pretty easy with just adding this command:
`
colcon build --cmake-args  -DCMAKE_BUILD_TYPE=Debug --merge-install
`

Should I just create a merge request for Garden release or change all of them one by one? Is there any easier method for changing documentation of all the releases?

chapulina, 2022-04-18:
There are a few tips for building with `colcon` here:

https://ignitionrobotics.org/docs/citadel/install_ubuntu_src#building-the-ignition-libraries

How about adding a note there?

> I think the right way is this but I will check that:

The syntax for combining multiple CMake args that has worked for me in the past is like this:

```
colcon build --cmake-args ' -DBUILD_TESTING=OFF' ' -DCMAKE_BUILD_TYPE=Debug'
```

I have a colcon cheatsheet [here](https://gist.github.com/chapulina/7400a708df655cbfba218e169fcad97f) if it helps.

> Should I just create a merge request for Garden release or change all of them one by one? Is there any easier method for changing documentation of all the releases?

One PR changing all of them is the best way we have now. You only need to update the currently supported / developed ones: Citadel, Fortress and Garden

onurtore, 2022-04-18:
Hi, 
Thanks for the -cmake-args tip. I also starred your colcon cheatsheet thats very helpful.

onurtore, 2022-04-20:
ping for a review.

closed 2022-05-23 by chapulina