gazebosim/docs #280: error in document for source install

lianghongzhuo, 2022-08-10:
For source installation the current instruction is to use the command:
```
sudo apt -y install \
  $(sort -u $(find . -iname 'packages-'`lsb_release -cs`'.apt' -o -iname 'packages.apt' | grep -v '/\.git/') | sed '/ignition\|sdf/d' | tr '\n' ' ')
```

However, it contains some packages that are not available or wrong.
Error msg:
```
E: Unable to locate package libgz-cmake3-dev
E: Unable to locate package libgz-common5-dev
E: Unable to locate package libgz-fuel-tools8-dev
E: Unable to locate package libgz-gui7-dev
E: Unable to locate package libgz-math7-dev
E: Unable to locate package libgz-math7-eigen3-dev
E: Unable to locate package libgz-msgs9-dev
E: Unable to locate package libgz-physics6-dev
E: Unable to locate package libgz-plugin2-dev
E: Unable to locate package libgz-rendering7-dev
E: Unable to locate package libgz-sensors7-dev
E: Unable to locate package libgz-sim7-dev
E: Unable to locate package libgz-tools2-dev
E: Unable to locate package libgz-transport12-dev
E: Unable to locate package libgz-utils2-cli-dev
E: Unable to locate package libgz-utils2-dev
E: Unable to locate package python3-gz-math7
```
Manually removing these packages works.

What I actually run:
```
sudo apt install binutils-dev build-essential cmake curl freeglut3-dev libassimp-dev libavcodec-dev libavdevice-dev libavformat-dev libavutil-dev libbenchmark-dev libcurl4-openssl-dev libdart-collision-bullet-dev libdart-collision-ode-dev libdart-dev libdart-external-ikfast-dev libdart-external-odelcpsolver-dev libdart-utils-urdf-dev libeigen3-dev libfreeimage-dev libgdal-dev libgflags-dev libglew-dev libgts-dev libjsoncpp-dev libogre-1.9-dev libogre-next-2.3-dev libogre-next-dev libprotobuf-dev libprotoc-dev libpython3-dev libsqlite3-dev libswscale-dev libtinyxml2-dev liburdfdom-dev libwebsockets-dev libxi-dev libxml2-utils libxmu-dev libyaml-dev libzip-dev libzmq3-dev pkg-config protobuf-compiler python3-dev python3-distutils python3-psutil python3-pybind11 python3-pytest python3-yaml qml-module-qtcharts qml-module-qtgraphicaleffects qml-module-qt-labs-folderlistmodel qml-module-qt-labs-platform qml-module-qt-labs-settings qml-module-qtlocation qml-module-qtpositioning qml-module-qtqml-models2 qml-module-qtquick2 qml-module-qtquick-controls qml-module-qtquick-controls2 qml-module-qtquick-dialogs qml-module-qtquick-layouts qml-module-qtquick-templates2 qml-module-qtquick-window2 qtbase5-dev qtdeclarative5-dev qtquickcontrols2-5-dev rubocop ruby ruby-dev swig uuid-dev xvfb
```
The above command successfully installed all the dependencies I need to compile gazebo from source.

chapulina, 2022-08-10:
This has been fixed in https://github.com/gazebosim/docs/pull/277, but we haven't deployed the updated tutorial to the site yet.

The trick is changing `ignition` to `gz`

lianghongzhuo, 2022-08-10:
Great! Thanks! Would be great if somebody update the documentation. :)

closed 2022-08-10 by lianghongzhuo