gazebosim/docs #331: force/torque senor example & how to add the link information to sensor plugin?

xibeisiber, 2023-01-30:
Hi, 

1. Is there an example on how to simulate the 6-axis force/torque sensor?

2. I put the robot model in one xacro file and intend to put ignition-related content in another xacro file.  In classic gazebo, there is a "bodyName" tag under the imu plugin tag. But in ignition, there is no thus tag in the tutorial. So the question is how to relate the sensor(say IMU) to certain link. 
I see in the example sensor_tutorial.sdf that the imu sensor tag should be under the link tag. But this way I have to the robot content and ignition content in the same file(sdf or xacro).

Thanks for any help!
Jia

xibeisiber, 2023-01-31:
1. I found the example file "force_torque.sdf" in gz_sim/test/worlds.
2. for the IMU, the sensor is bound to a link, so I can use `<gazebo reference="base_link">` in the ignition-related xacro file.
But the force-torque sensor is bound to a joint, the `<gazebo reference="jointname">` seems not working. 
file:///opt/ros/humble/share/ros_gz_sim_demos/models/rrbot.xacro

xibeisiber, 2023-02-02:
Partly solved.
For some unknown reason, the `<plugin filename="ignition-gazebo-forcetorque-system" name="ignition::gazebo::systems::ForceTorque" />` can not be successfully loaded when written in xacro file.
A makeshift solution is to add this plugin in `/usr/share/ignition/ignition-gazebo6/worlds/empty.sdf`

azeey, 2023-02-06:
The reason is the force torque sensor system, like the IMU sensor system, is meant to be loaded as part of the world as it handles all the force torque sensors in the world, not just a single model. I think the IMU sensor works when loaded as a model plugin, but it's just a side-effect of how it's implemented. Users are not meant to load it as a model plugin.

For your ROS2 usage, I recommend you copy the `empty.sdf` into your own package, modify it to include the `ForceTorque` sensor under `<world>` and use that in your launch file with the `gz_args` argument of `gz_sim.launch.py`.

closed 2023-02-06 by azeey

Nibanovic, 2024-01-10:
**tldr:** My two cents: to work around this issue, you don't need to define the whole sensor in `empty.sdf`, only the plugin.

I just struggled with the same issue. For me, I only had to add `ForceTorque` plugin under the `<world name="empty">` tag, as so:
```
  <world name="empty">
    <physics name="1ms" type="ignored">
      <max_step_size>0.001</max_step_size>
      <real_time_factor>1.0</real_time_factor>
    </physics>
### ... some other plugins
    <plugin 
      filename="ignition-gazebo-forcetorque-system"
      name="ignition::gazebo::systems::ForceTorque">
    <plugin>
```

Then, the sensor defined in my `URDF` was being properly read:

```
      <gazebo reference="${prefix}joint5">
        <sensor name="force_torque_sensor" type="force_torque">
          <always_on>true</always_on>
          <update_rate>100</update_rate>
          <visualize>true</visualize>
          <topic>force_torque</topic>
          <enable_metrics>true</enable_metrics>
          <pose>0 0 0 0 0 0</pose>
          <force_torque>
            <frame>sensor</frame>
            <measure_direction>child_to_parent</measure_direction>
          </force_torque> 
        </sensor>
      </gazebo>
```

jaronski, 2024-05-29:
i am trying to get ros2_control force_torque_sensor_broadcaster to publish the data from the force torque sensor from inside of ign fortress. when i try to start the ros2 control load_controller fts_sensor_broadcaster --set-state active it shows green and "as if" its working but when i list_controllers it tells me the fts_sensor_broadcaster is inactive. 

`[ign-1] [ERROR] [1716986934.994304661] [controller_manager]: Can't activate controller 'ftsensor_broadcaster': State interface with key 'sensor_FT/force.x' does not exist
[spawner-8] [INFO] [1716986935.005369509] [spawner_ftsensor_broadcaster]: Configured and activated ftsensor_broadcaster`

i tried renaming the sensor, state_interfaces in the config.yaml and xacro of my custom robot model. I added the plugin as described here into the empty_world.sdf that i launch 
```
<!-- FT Plugin -->
    <plugin filename="ignition-gazebo-forcetorque-system" name="ignition::gazebo::systems::ForceTorque"/>
```
is this working at all? i tried also looking into the universal robots repos but i guess the force torque part is not working as i couldn't find a ur_controller/force_torque_sensor_broadcaster as stated in the config.yaml

Does anyone here have a solution for this or successfully integrated the Force Torque Sensor in Ignition Fortress and ros2 humble ros2_control ?

Thanks for any hints or tips

SteMuc, 2024-09-28:
Hi, I was facing the same issue with the FT sensor plugins but I was able to solve by following you guidelines.

I added the ```
<?xml version="1.0"?>

<sdf version="1.6">
  <world name="ft_world">
    <physics name="1ms" type="ignored">
      <max_step_size>0.001</max_step_size>
      <real_time_factor>1.0</real_time_factor>
    </physics>

    <plugin filename="ignition-gazebo-physics-system" name="ignition::gazebo::systems::Physics">
    </plugin>
    <plugin filename="ignition-gazebo-user-commands-system" name="ignition::gazebo::systems::UserCommands">
    </plugin>
    <plugin filename="ignition-gazebo-scene-broadcaster-system" name="ignition::gazebo::systems::SceneBroadcaster">
    </plugin>
    <plugin filename="ignition-gazebo-contact-system" name="ignition::gazebo::systems::Contact">
    </plugin>
    <plugin filename="ignition-gazebo-forcetorque-system" name="ignition::gazebo::systems::ForceTorque">
    </plugin>

    <light type="directional" name="sun">
      <cast_shadows>true</cast_shadows>
      <visual>false</visual>
      <pose>0 0 100 0 0 0</pose>
      <diffuse>0.8 0.8 0.8 1</diffuse>
      <specular>0.2 0.2 0.2 1</specular>
      <attenuation>
        <range>1000</range>
        <constant>0.9</constant>
        <linear>0.01</linear>
        <quadratic>0.001</quadratic>
      </attenuation>
      <direction>-0.5 0.1 -0.9</direction>
    </light>

    <model name='ground_plane'>
      <static>true</static>
      <link name='link'>
        <collision name='collision'>
          <geometry>
            <plane>
              <normal>0 0 1</normal>
              <size>100 100</size>
            </plane>
          </geometry>
          <surface>
            <friction>
              <ode />
            </friction>
            <bounce />
            <contact />
          </surface>
        </collision>
        <visual name='visual'>
          <geometry>
            <plane>
              <normal>0 0 1</normal>
              <size>100 100</size>
            </plane>
          </geometry>
          <material>
            <ambient>0.8 0.8 0.8 1</ambient>
            <diffuse>0.8 0.8 0.8 1</diffuse>
            <specular>0.8 0.8 0.8 1</specular>
          </material>
        </visual>
        <pose>0 0 0 0 -0 0</pose>
        <inertial>
          <pose>0 0 0 0 -0 0</pose>
          <mass>1</mass>
          <inertia>
            <ixx>1</ixx>
            <ixy>0</ixy>
            <ixz>0</ixz>
            <iyy>1</iyy>
            <iyz>0</iyz>
            <izz>1</izz>
          </inertia>
        </inertial>
        <enable_wind>false</enable_wind>
      </link>
      <pose>0 0 0 0 -0 0</pose>
      <self_collide>false</self_collide>
    </model>

  </world>
</sdf>
``` with the FT and then inside the URDF I added the following sensor:      ```
 <gazebo reference="${tf_prefix}wrist_3_joint">
        <sensor name="force_torque_sensor" type="force_torque">
          <always_on>true</always_on>
          <update_rate>50</update_rate>
          <visualize>true</visualize>
          <topic>force_torque</topic>
          <force_torque>
            <frame>child</frame>
            <measure_direction>child_to_parent</measure_direction>
            
            
          </force_torque>
        </sensor>
      </gazebo>
``` The problem with this kind of plugin is that it does not work with a type joint fixed, but only revolute.

Thank you again for you response!

Best regards,

Stefano

Nithish-Kumar19181, 2025-10-08:
<gazebo reference="${tf_prefix}wrist_3_joint">
       <sensor name="force_torque_sensor" type="force_torque">
         <always_on>true</always_on>
         <update_rate>50</update_rate>
         <visualize>true</visualize>
         <topic>force_torque</topic>
         <force_torque>
           <frame>child</frame>
           <measure_direction>child_to_parent</measure_direction>
           
           
         </force_torque>
       </sensor>
     </gazebo>

by using this i can see the force torque sensor working in gazebo but to what topic are the sensor values being published ?? 
because when i run ros2 topic list im unable to see the topic force_torqe
ros2 topic list 
/clicked_point
/clock
/dynamic_joint_states
/goal_pose
/ik_trajectory_markers
/initialpose
/joint_state_broadcaster/transition_event
/joint_states
/joint_trajectory_controller/controller_state
/joint_trajectory_controller/joint_trajectory
/joint_trajectory_controller/state
/joint_trajectory_controller/transition_event
/parameter_events
/performance_metrics
/robot_description
/rosout
/target_joint_positions
/tf
/tf_static