gazebosim/docs #334: Add tutorial for using DEM's for gazebo garden

Ryanf55, 2023-02-02:
Unfortunately, the only tutorial available is from gazebo-classic found here: https://classic.gazebosim.org/tutorials?tut=dem&cat=build_world

It is outdated but will hopefully get you in the right direction. When following the tutorial, you should be able to copy this section into a Gazebo Garden compatible `SDF` world: https://github.com/gazebosim/gazebo-classic/blob/1bef2dc573aa15ca8a56b4a7dadb9905cf978582/worlds/heightmap_dem.world#L8-L54

If you'd like to contribute, feel free to open a PR for a newer tutorial on DEMs here: https://github.com/gazebosim/docs :smile:

_Originally posted by @jennuine in https://github.com/gazebosim/gz-sim/issues/235#issuecomment-1412848224_
            

Creating an issue to keep track of this request. Once I free some time up, I will try the suggestions above and if it works, I'll be happy to submit a PR.

Ryanf55, 2023-02-02:
Ok, using ubuntu 22.04, following the gazebo classic tutorial. 

1. `sudo apt-get install gdal-bin libgdal-dev python-gdal` fails because
```
Package python-gdal is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  gdal-bin
```
Looking `gdal` up on [ubuntu's jammy listings](https://packages.ubuntu.com/search?suite=jammy&keywords=gdal) shows the following packages. `sudo apt-get install gdal-bin  libgdal-dev python3-gdal` 
2. gdalwarp  to reduce resolution worked fine
3. `vi /tmp/volcano.world` to create the volcano file. 
4. I tried running it on garden like so, but the URI's for the models don't resolve. 
```
ubuntu@ubuntu22:~/Downloads$ gz sim /tmp/volcano.world 

Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.

[Err] [Server.cc:159] Error Code 14: [/sdf/world[@name="default"]/include[0]/uri:/tmp/volcano.world:L6]: Msg: Unable to find uri[model://sun]
```
5. Next, trying to follow the [tutorial for creating a model](https://gazebosim.org/docs/garden/building_robot), I can't get it to run, likely due to some restraints of running in virtualbox. Until I get my new computer that can run OpenGL3.3 native, I will have to put this work on hold. Anyone else is welcome to continue from where I left off.

jennuine, 2023-02-02:
Are you able to run `gz sim volcano.sdf` ok? `volcano.sdf` is an example that comes packaged with Garden. If so, then it is most likely an issue with the world file. `.world` files were for gazebo-classic whereas `.sdf` is used for the new Gazebo. If you're not able to run the `volcano.sdf` example, then it may have something to do with using a virtual machine

Ryanf55, 2023-02-03:
No, I am not. Can you point me to which repo I would find it in?
```
ubuntu@ubuntu22:~$ gz sim volcano.sdf
Unable to find or download file 
```

jennuine, 2023-02-03:
https://github.com/gazebosim/gz-sim/blob/gz-sim7/examples/worlds/dem_volcano.sdf

Ryanf55, 2023-02-03:
> https://github.com/gazebosim/gz-sim/blob/gz-sim7/examples/worlds/dem_volcano.sdf

Thanks that worked. I should be able to replicate it into the guide. For now, I will leave out the DEM generation if that's ok. 
[image]

jennuine, 2023-02-03:
> For now, I will leave out the DEM generation if that's ok.

Yups, sounds good. Thanks for tackling this!

Ryanf55, 2023-02-06:
Where do the other files in the original `.world` come from, such as: `file://media/materials/textures/dirt_diffusespecular.png`? There is no information on where to retrieve those. 

I was able to load in the DEM file by using the new `GZ_SIM_RESOURCE_PATH` variable.

jennuine, 2023-02-06:
The original files are located [here in the gazebo-classic repo](https://github.com/gazebosim/gazebo-classic/tree/gazebo11/media/materials/textures) but I think you should point users to these files in gz-rendering: https://github.com/gazebosim/gz-rendering/tree/gz-rendering7/examples/heightmap/media

For example, in the tutorial it can say something like "download these xyz files from gz-rendering"

Ryanf55, 2024-01-23:
I have a computer capable of running much higher fidelity simulations now, including terrain data, without crashing. I'll try to get this before November's release.

mabelzhang, 2024-01-23:
By November, you mean September? Like, Ionic release? Or is this talking about something else?

Ryanf55, 2024-01-23:
> By November, you mean September? Like, Ionic release? Or is this talking about something else?

Ionic release. Let me know a date it needs done, and I'll hold myself to it:)

mabelzhang, 2024-01-23:
Here are the release dates https://github.com/gazebosim/docs/blob/master/releases.md
But the feature freeze date is much earlier than the release date. I don't know if we decided freeze dates for Ionic yet. For reference, Harmonic feature freeze was Jul 28, 2023, code freeze Aug 30, and release was end of Sep.
So I'd aim somewhere in July to be safe...

Ryanf55, 2024-04-20:
We should also link this: https://gazebosim.org/api/rendering/8/heightmap.html

jennuine, 2024-04-22:
> The statement the example 1 demo, loads the heights from a grayscale image, where the color black represents the lowest point and white represents the heights.  does not seem accurate though

Hello @Ryanf55  👋  Why does this statement not appear accurate? I will admit, the formatting of the tutorial may be a bit confusing. There are two imaged examples, the first one presents a image heightmap (using grayscale data) and the second one below the first demos a DEM heightmap

Ryanf55, 2024-04-23:
Yea, just some weird wording. I got a little lost on that, and confused the color references being the color of the outputted visuals. 
Perhaps we could change it to talk about how the blending works, and maybe choose opposing colors such as orange and blue instead of adding in textures for a starter tutorial?

jennuine, 2024-04-23:
PRs are welcome to help clarify.

> maybe choose opposing colors such as orange and blue

I would argue that white and black are opposing colors. Also, in terms of the image heightmap, it's not necessarily about the colors but how the code reads these scalar values to determine height information. If we take a closer look at the bowl heightmap example, the black represents the lowest point on the map 0 and white represents the highest 255. Anything in between is scaled appropriately. Essentially it reads the grayscale image and outputs the corresponding heightmap (bottom image).

[image]

[image]

usmanNoor5, 2024-08-09:
how to insert different objects in gazebo harmonic

jennuine, 2024-08-09:
> how to insert different objects in gazebo harmonic

This doesn't seem to be related to this ticket but you can visit our tutorials page for more details on inserting models either building your own or from fuel: https://gazebosim.org/docs/latest/tutorials/#

Another resource to browse and ask questions is https://community.gazebosim.org/