gazebosim/docs #355: Broken links on https://gazebosim.org/libs

peci1, 2023-03-27:
Most "API & Tutorials" links are dead on https://gazebosim.org/libs .

I've noticed:

- https://gazebosim.org/api/common/5.3/index.html is dead
- https://gazebosim.org/api/common/5/index.html is good

So maybe this page just needs to stop using the minor version in the links?

azeey, 2023-03-30:
@mabelzhang I think you also noticed this right?

mabelzhang, 2023-03-30:
Yep

traversaro, 2023-05-10:
> So maybe this page just needs to stop using the minor version in the links?

I noticed that some libraries work (see https://gazebosim.org/api/fuel_tools/8.0/index.html), but actually link to older version of the documentation instead of the latest one available at https://gazebosim.org/api/fuel_tools/8/index.html .

peci1, 2023-09-02:
This issue still persists.

EyalBrilling, 2023-12-12:
Where "https://gazebosim.org/libs" sits? I couldn't find it in this repository(guess because it isn't under gazebosim.org/docs) but quick going over other repositories i didn't find it either.

azeey, 2023-12-12:
The website is being reworked so it's statically generated, so this will change, but currently it's a being served by the [libs](https://github.com/gazebo-web/gazebosim-web-frontend/blob/main/src/app/libs/libs.component.html) component of the Angular app.

azeey, 2024-07-18:
Fixed by #441.

closed 2024-07-18 by azeey