gazebosim/docs #359: Left menu on Docs page shows "Ros/gazebo Installation" instead of "ROS/Gazebo Installation"

Bi0T1N, 2023-04-27:
## Description
* Expected behavior: Menu entry should be **ROS/Gazebo Installation**
* Actual behavior: Menu entry is **Ros/gazebo Installation**

## Steps to reproduce
1. Go to https://gazebosim.org/docs
2. Notice that the left menu shows "[Ros/gazebo Installation](https://gazebosim.org/docs/garden/ros_installation)"
3. Compare with what is written in [index.yaml](https://github.com/gazebosim/docs/blob/master/index.yaml#L23)

Bi0T1N, 2023-04-27:
It's not just the single page listed above, it's wrong for all entries with more than one word. To give some more examples:
- `Ros 2 Integration` instead of `ROS 2 Integration`
- `Spawn Urdf` instead of `Spawn URDF`
- `Understanding The Gui` instead of `Understanding the GUI`

It uses its own mechanism for capitalization of the output instead of respecting what is defined in the `index.yaml` file.

Bi0T1N, 2023-05-06:
I'm not an expert in Golang but the casing might come from the [`strings.Title()`](https://pkg.go.dev/strings#Title) function. It's not used in [gazebosim-web-backend](https://github.com/gazebo-web/gazebosim-web-backend) or [gz-go](https://github.com/gazebo-web/gz-go) but within the [`go-gorm`](https://github.com/go-gorm/gorm/blob/e61b98d69677b8871d832baf2489942d79054a4a/schema/naming.go#LL111C99-L111C99) dependency.
At least the following examples show the same phenomenon that can be observed at the moment:
```go
package main

import (
	"fmt"
	"strings"
)

func main() {
	fmt.Println(strings.Title(strings.ToLower("Migration from Ignition"))) // Migration From Ignition
	fmt.Println(strings.Title(strings.ToLower("Spawn URDF"))) // Spawn Urdf
	fmt.Println(strings.Title(strings.ToLower("Understanding the GUI"))) // Understanding The Gui
}
```

azeey, 2024-07-18:
Fixed by #441

closed 2024-07-18 by azeey