gazebosim/docs #384: Tutorial layout has issues with narrow Windows

mjcarroll, 2023-09-11:
I noticed that when resizing a window with the tutorial open, that when the window width is less than approximately ~1000 pixels, then presentation gets a little wonky.

The right side table of contents will start to flow over the edit pencil and the content of the tutorial

[image]

When the window gets sufficiently narrow, the content will overflow the window: 
[image].

This makes it difficult to have the tutorial window and the gazebo window/terminal open side-by-side on smaller displays.

mabelzhang, 2023-09-18:
Should this ticket be in https://github.com/gazebo-web/gazebosim-web-frontend/issues?
Also is the frontend repo automated for the project board?

azeey, 2024-07-18:
Fixed by #441 
[Screen recording 2024-07-18 13.20.41.webm](https://github.com/user-attachments/assets/3ce22cd4-47b7-4ca0-9911-f91766ad4739)

closed 2024-07-18 by azeey