gazebosim/docs #397: Garden install instructions hang on dependency installation command

tfoote, 2023-10-02:
Originally reported here: https://community.gazebosim.org/t/unable-to-install-gazebo-garden-in-ubuntu-20-04/2296

It appears that in https://gazebosim.org/docs/garden/install_ubuntu_src

Source: 

https://github.com/gazebosim/docs/blob/master/garden/install_ubuntu_src.md#install-dependencies

The `$(find . -iname 'packages-'`lsb_release -cs`'.apt' -o -iname 'packages.apt')` returns empty. Which then the `sort -u` will hang waiting for an input. 

I believe that there's an unwritten assumption about what path you're executing in. You need to be in the directory of the workspace with the checked out code.

scpeters, 2023-10-02:
I've opened a draft PR to clarify the working directory needed for this step: https://github.com/gazebosim/docs/pull/398

scpeters, 2023-10-03:
fixed by #398

scpeters, 2023-10-03:
the change hasn't yet deployed to gazebosim.org

closed 2023-10-03 by scpeters