gazebosim/docs #408: Feature Request: Add documentation for SDF file format across all sensors.

ShotaAk, 2023-11-15:
## Desired behavior

**Current Situation:**
Gazebo official tutorials cover IMU and LiDAR usage, but lack documentation for other sensors like cameras and FT sensors.

https://gazebosim.org/docs/harmonic/sensors

[image]

**Proposal:**
Please expand the official tutorials to include guides on integrating various sensors (e.g., cameras, FT sensors). Include details on plugin usage and parameter configurations for each sensor type.

## Alternatives considered

If documentation for all sensors' plugins and parameters exists, please add links to this information in the official tutorials. 

## Implementation suggestion

Table of contents: 

- Ignition launch
- Video walk-through
- **Other sensors**
  - **RGB Camera**
  - **Depth Camera**
  - **Force Torque Sensor**
  - **Air Pressure Sensor**
 

Force Torque Sensor page:

world file:
```xml
<?xml version="1.0" ?>
<sdf version="1.6">
  <world name="default">
    <plugin
      filename="ignition-gazebo-forcetorque-system"
      name="ignition::gazebo::systems::ForceTorque">
    </plugin>
  </world>
</sdf>
```

model file (URDF):

```xml
<?xml version="1.0"?>
<robot xmlns:xacro="http://ros.org/wiki/xacro">
    <gazebo reference="joint1">
      <sensor name="force_torque_sensor" type="force_torque">
        <update_rate>100.0</update_rate>
        <always_on>1</always_on>
        <topic>joint1/force_torque</topic>
        <visualize>false</visualize>
        <ignition_frame_id>link1</ignition_frame_id>
        <force_torque>
          <frame>child</frame>
          <measure_direction>child_to_parent</measure_direction>
        </force_torque>
      </sensor>
    </gazebo>
  </xacro:macro>
</robot>
```

## Additional context

Ryanf55, 2024-01-23:
If the RGB camera API can't be given in a tutorial, perhaps a link to the header file could be added?

mabelzhang, 2024-01-23:
The example worlds are helpful too: https://github.com/gazebosim/gz-sim/tree/gz-sim8/examples/worlds
e.g. camera_sensor https://github.com/gazebosim/gz-sim/blob/gz-sim8/examples/worlds/camera_sensor.sdf

A tutorial could start from the existing example worlds, and not have to start from scratch.