gazebosim/docs #424: GUI camera doesn't keep manual orientation changes when in Follow entity mode

NDevDrone, 2024-02-01:
## Environment
* OS Version: Ubuntu 22.04
* Source or binary build?
Garden, both source and binary. 

## Description
* Expected behavior: Rotating camera in follow mode should change orientation of the tracking camera.
* Actual behavior: The rotation changes but then returns to the original follow orientation. 
## Steps to reproduce

1. Follow entity.
2. Rotate or pan GUI camera view.
3. Sit in despair as you watch the camera rotate back to its default, as you can't find a means to lock new camera orientation while following anywhere within the GUI. 

## Output
N/A

NDevDrone, 2024-02-01:
@srmainwaring , would you happen to have worked out a solution to this?

srmainwaring, 2024-02-01:
@NDevDrone there is a service for the follow camera where you can reposition and reorient the camera: For example:

```bash
gz service -s /gui/follow/offset --reqtype gz.msgs.Vector3d --reptype gz.msgs.Boolean --timeout 2000 --req "x: -0.5, y: 0.5, z: 0.5"
```

The camera controller struggles to keep the image smooth when close to a fast moving vehicle, there is a service for adjusting that as well:

- https://github.com/gazebosim/gz-gui/pull/515