gazebosim/docs #449: Installing Gazebo11 side by side with new Gazebo not working

pariaspe, 2024-06-04:
## Environment
* OS Version: Ubuntu 22.04, on docker from `osrf/ros:humble-desktop`
* Source or binary build? Binary `gazebo11` and `gz-harmonic`

## Description
* Expected behavior: classic Gazebo and Gazebo Harmonic installed together
* Actual behavior: Only Gz Harmonic is installed

## Steps to reproduce
After following the steps of [Installing Gazebo11 side by side with new Gazebo](https://github.com/gazebosim/docs/blob/9383b6c4d0943772703c754a10b4cedddcab3db1/install_gz11_side_by_side.md), only Gz Harmonic is installed.

Tested on docker with the following Dockerfile:
```Dockerfile
FROM osrf/ros:humble-desktop

# Install common tools
RUN apt-get update && apt-get install -y \
    lsb-release wget gnupg software-properties-common \
    && rm -rf /var/lib/apt/lists/* 

# Install gazebo11
RUN add-apt-repository ppa:openrobotics/gazebo11-gz-cli
RUN apt-get update && apt-get install -y \
    gazebo11 \
    && rm -rf /var/lib/apt/lists/* 

      # Install Gazebo Harmonic
RUN apt-get update \
    && wget https://packages.osrfoundation.org/gazebo.gpg -O /usr/share/keyrings/pkgs-osrf-archive-keyring.gpg \
    && echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/pkgs-osrf-archive-keyring.gpg] http://packages.osrfoundation.org/gazebo/ubuntu-stable $(lsb_release -cs) main" | tee /etc/apt/sources.list.d/gazebo-stable.list > /dev/null \
    && apt-get update && apt-get install -y -q \
        gz-harmonic \
    && rm -rf /var/lib/apt/lists/*

ENTRYPOINT [ "/bin/bash" ]
```

What am I missing?

## Output
[image]

yaswanth1701, 2024-06-04:
Hi @j-rivero this is the issue which I mentioned yesterday in the weekly meet.

j-rivero, 2024-06-06:
Confirmed the problem is there. I think that the problem is the version of gz-tools2 package that needs to be exactly `2.0.1-2` and I released it for Noble but not for Jammy. The packages should be now in the repo, could you please give it another try using the `--no-cache` option?

pariaspe, 2024-06-07:
It's working now, thanks guys!

[image]

closed 2024-06-07 by pariaspe

pariaspe, 2024-06-11:
Hi @yaswanth1701 @j-rivero , sorry to bother you guys. Are [gazebo_ros_pkgs](https://index.ros.org/r/gazebo_ros_pkgs/github-ros-simulation-gazebo_ros_pkgs) also available?

```
6.703 Some packages could not be installed. This may mean that you have
6.703 requested an impossible situation or if you are using the unstable
6.703 distribution that some required packages have not yet been created
6.703 or been moved out of Incoming.
6.703 The following information may help to resolve the situation:
6.703 
6.703 The following packages have unmet dependencies:
6.747  libgazebo-dev : Depends: libgazebo11 (= 11.10.2+dfsg-1) but 11.14.0-1~alpha12 is to be installed
6.747                  Depends: gazebo-common (= 11.10.2+dfsg-1) but it is not installable
6.750 E: Unable to correct problems, you have held broken packages.
------
Dockerfile-gz:10
--------------------
   9 |     RUN add-apt-repository ppa:openrobotics/gazebo11-gz-cli
  10 | >>> RUN apt-get update && apt-get install -y \
  11 | >>>     gazebo11 \
  12 | >>>     ros-humble-gazebo-ros-pkgs \
  13 | >>>     && rm -rf /var/lib/apt/lists/* 
  14 |     
--------------------
```

yaswanth1701, 2024-06-11:
Hi @pariaspe, I am using Jammy and did the binary installation of Gazebo Classic (side-by-side installation) and source installation of Gazebo Harmonic. I can use Gazebo Classic from the terminal as described in the installation tutorial but I faced the same issue with libgazebo11-dev installation. I thought it might be due to jammy but It seems it's the case with noble too.

yaswanth1701, 2024-06-11:
ig Jose can tell us about this.

j-rivero, 2024-06-24:
> ig Jose can tell us about this.

Working on the fix. Let's follow https://github.com/gazebo-tooling/release-tools/issues/1152