gazebosim/docs #483: Windows instructions should be updated

azeey, 2024-08-13:
As of https://github.com/gazebosim/gz-sim/pull/2382, Gazebo actually runs with Ogre2 on windows, but our [documentation](https://gazebosim.org/docs/latest/install_windows_src/) states that only Ogre (1) is supported. The ["GUI workaround"](https://gazebosim.org/docs/latest/install_windows_src/#gazebo-gui-workaround
) section also mentions that `gz sim -g` does not work.

The [binary](https://gazebosim.org/docs/latest/install_windows/) install instructions also say:
> Additionally, command line tools, the DART physics engine, and some tests are not currently supported in Windows.

which is no longer accurate.

azeey, 2024-08-13:
@traversaro any help with this would be appreciated.

akky20, 2024-08-16:
@azeey I have made some changes according to the mentioned point , please have a look . I’m willing to accept any corrections.

1
 [image]

2
 [image]

3
 [image]

  Removed the line "Additionally, command line tools..."

azeey, 2024-08-16:
@akky20 I think would be best if you to go ahead and create a PR with what you have and I'll give my feedback there.

akky20, 2024-08-16:
> @akky20 I think would be best if you to go ahead and create a PR with what you have and I'll give my feedback there.

Opened the PR , please have a look

PR #487 (2024-08-29): Windows instructions updated in harmonic branch

# 🦟 Bug fix

Fixes #483 

## Summary
Changes doc text according to the mentioned parts in issue

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/all/contributing#test-coverage))
- [ ] While waiting for a rev [... cut]
closed 2024-08-29 by azeey