gazebosim/docs #507: Windows build-from-source instructions are confusing with respect to Condaforge

clalancette, 2024-09-11:
## Environment
* OS Version: Windows 10
* Source or binary build?  Source, Ionic (gz-cmake4, gz-common6, gz-fuel-tools10, gz-gui9, gz-launch8, gz-math8, gz-msgs11, gz-physics3, gz-rendering9, gz-sensors9, gz-sim9, gz-tools2, gz-transport14, gz-utils3, sdf15)

## Description

Installation of conda forge, as the first step of https://gazebosim.org/docs/ionic/install_windows_src/ , is confusing to people who aren't familiar with conda and the Windows ecosystem.  My suggestion is to make an opinionated choice on which version of conda to install, and with which options.  A single-line command to install conda would be ideal.

azeey, 2024-09-11:
@traversaro can we get your help with this?

traversaro, 2024-09-11:
> @traversaro can we get your help with this?

Sure, please assign it to me so I remember to do this (I am currently travelling). Thanks!

traversaro, 2024-09-14:
I tried to simplify the installation process. I could not get a single line of commands to install, but hopefully the commands are much more clear now.

closed 2024-09-16 by ahcorde