gazebosim/docs #517: Windows install instructions do not provide python3.exe, only python.exe

peci1, 2024-09-15:
## Environment
* OS Version: Win 10
* Source or binary build? Source, Ionic prerelease

## Description
* Expected behavior: All tutorials use `python3` command. I should be able to follow such tutorials even on Windows.
* Actual behavior: The conda-installed python only provides `python`. When I call `python3`, it bubbles up to the Windows py.exe launcher which opens Windows Store.

## Steps to reproduce

1. Install gazebo following the official Win from source installation guide.
2. Type `python3` in console.

## Relevant info

There are a few ways to provide the `python3` command, but I don't know which one is the most suitable: https://superuser.com/a/1576801 .

azeey, 2024-09-16:
Any thoughts @traversaro?

traversaro, 2024-09-16:
> Any thoughts @traversaro?

This is a tough one unfortunately. This is not a conda-specific issue, but rather a generic problem of both the official Windows distribution of python and the conda-forge distribution, see:
* Official python issue: https://github.com/python/cpython/issues/99185
* conda-forge issue: https://github.com/conda-forge/python-feedstock/issues/349

TL;DR: No official python3.exe will be shipped anytime soon either in conda-forge or in the official Windows python binaries.

Personally I typically use `python` and document that apt users should install `python-is-python3`, but that is not a suitable solution here, so we should provide some kind of alias here, but ideally in a way that is environment specific. Perhaps we can just document to add a `python3.bat` path in the colcon's workspace `./bin` folder with content `python.exe %*` ?

peci1, 2024-09-16:
Outch. That's quite unfortunate.

> Perhaps we can just document to add a `python3.bat` path in the colcon's workspace `./bin` folder with content `python.exe %*` ?

I was thinking about something like that. The file could even be added to (which?) on of the gz repos so that users can just download it and place it in the bin folder.

traversaro, 2024-09-16:
> I was thinking about something like that. The file could even be added to (which?) on of the gz repos so that users can just download it and place it in the bin folder.

Probably the most root dependency that has python bindings, so gz-math? Ideally that could be disabled by a CMake option.

traversaro, 2024-09-17:
Otherwise, we can just keep simple (and a bit ugly) and document that on Windows `python.exe` should be used instead of `python3`, as anyhow somebody working with python on Windows need to be aware of that.

azeey, 2024-09-17:
> Otherwise, we can just keep simple (and a bit ugly) and document that on Windows python.exe should be used instead of python3, as anyhow somebody working with python on Windows need to be aware of that.

I'd be in favor of that. Solving this by other means than documentation seems to be out of scope for Gazebo.

peci1, 2024-09-19:
It seems to me the whole collection would profit from a document summarizing what do Windows users need to do differently to successfully follow the tutorials. There's more things than python:

1. Replace trailing `\` with `^` in commands (or better remove them completely) (not sure what PowerShell needs here)
2. Use python instead of python3
3. Run `gz sim -s` and `gz sim -g` instead of a single `gz-sim` command. It should also tell whether all arguments should go to both commands, or how to divide them.
4. Add `--render-engine ogre` to some commands to get at least some rendering.
5. More...?

At the beginning of every tutorial, there could be a link to this document. It could be something quite non-intrusive, like

> If you're using Windows CMD or PowerShell, the tutorial commands need slight adjustment &lt;link&gt;.

I'm not sure whether Mac would also profit from something like that.

The very ideal solution would be doc pages with multiple fragments for each code block, one for each OS that needs to do something differently. And very very ideally, the correct code block would be selected by platform detection. However, if the percentage of Windows users is small, the effort might not be worth it (it would slow down docs contributions from people who do not want to care about Windows).