gazebosim/docs #558: Tutorial launch file fails woth missing package `spaceros_gz_sim`

AndyBlightLeeds, 2025-01-08:
## Environment
* OS Version: 
Ubuntu 24.04 LTS
* Source or binary build:
Binary: Gazebo Sim, version 8.7.0 (Harmonic LTS)

## Description
* Expected behavior: 
I was working through the tutorial on this page: <https://gazebosim.org/docs/harmonic/ros2_launch_gazebo/>.  When I ran the first Python launch file on this page, I got an error (see below).   I expected to be able to start Gazebo and see some sort of working demo.

* Actual behavior: 
When I ran the launch file, I got the following error: 

```bash
$ ros2 launch ./gazebo.launch.py 
[INFO] [launch]: All log files can be found below /home/...
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught multiple exceptions when trying to load file of format [py]:
 - PackageNotFoundError: "package 'spaceros_gz_sim' not found, searching: ['/opt/ros/jazzy']"
 - InvalidFrontendLaunchFileError: The launch file may have a syntax error, or its format is unknown
```

This looks like I need to install another package called `spaceros_gz_sim` but there is nothing in the documentation that I could find that tells me what to install. 

I checked the later versions of the tutorials (`Ionic` and `Jetty`) for any fixes but they both reference `spaceros_gz_sim` which I don't have.

Please provide a simpler launch file that works with no additional packages or add notes on what packages to install to make the launch file work correctly. 

## Steps to reproduce
1. Install Gazebo Sim (I used Harmonic with ROS extensions).
2. Navigate to the page: <https://gazebosim.org/docs/harmonic/ros2_launch_gazebo/>
3. Copy and paste the first Python launch file code into a file `gazebo.launch.py` 
4. Launch the file using `ros2 launch ./gazebo.launch.py`
5. The error message is seen. 

## Output
```bash
$ gz sim --version
Gazebo Sim, version 8.7.0
Copyright (C) 2018 Open Source Robotics Foundation.
Released under the Apache 2.0 License.
```

```bash
$ ros2 launch ./gazebo.launch.py 
[INFO] [launch]: All log files can be found below /home/...
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught multiple exceptions when trying to load file of format [py]:
 - PackageNotFoundError: "package 'spaceros_gz_sim' not found, searching: ['/opt/ros/jazzy']"
 - InvalidFrontendLaunchFileError: The launch file may have a syntax error, or its format is unknown
```

azeey, 2025-01-08:
Ah, I don't think we should have a dependency on `spaceros_gz_sim` in that tutorial. It was added in https://github.com/gazebosim/docs/pull/482. @david-dorf would you mind updating the tutorial so that it works without a dependency on `spaceros_gz_sim`?

AndyBlightLeeds, 2025-01-08:
I have just checked the Python server tutorial on the same page and this fails like this:

```bash
$ ros2 launch src/gazebo_test/gazebo_server.launch.py 
[INFO] [launch]: All log files can be found below /home/...
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): launch configuration 'world_sdf_file' does not exist
```

Can this be fixed at the same time or do I need to create another issue?

azeey, 2025-01-08:
> I have just checked the Python server tutorial on the same page and this fails like this:

That's saying you need to pass the `world_sdf_file` argument when you run it from the command line. 
Something like `ros2 launch ros_gz_sim ros_gz_sim.launch.py world_sdf_file:=empty.sdf`.
The example invocation is in https://gazebosim.org/docs/harmonic/ros2_launch_gazebo/#launching-with-ros-gz-bridge.

david-dorf, 2025-01-08:
> Ah, I don't think we should have a dependency on `spaceros_gz_sim` in that tutorial. It was added in https://github.com/gazebosim/docs/pull/482. @david-dorf would you mind updating the tutorial so that it works without a dependency on `spaceros_gz_sim`?

Sure, I'll update it to normal gz today, apologies for that

david-dorf, 2025-01-08:
Made a pr for a more generic example, but probably won't run on its own because it requires the user to replace the "example..." parts with their own (such as the world, package, bridged topics, etc.). Maybe there's an existing world that could be used instead though so it runs as a standalone as well?

AndyBlightLeeds, 2025-01-09:
@azeey Thanks for pointing out the way to set the world from the command line.  

@david-dorf The pages have been updated already and after a bit of customisation of the example script, I was able to run the Gazebo with a ROS2 topic being published.  I used the string `empty.sdf` for the world value, so this could be used in the example.

Thanks to both of you for your help, especially the speedy responses. 

This issue can be closed now.

closed 2025-01-09 by azeey