gazebosim/docs #572: Alert and link to newer library versions for API like ros2 docs

EmmanuelMess, 2025-02-11:
Gazebo docs doesnt link to newer versions for its API. See for example https://gazebosim.org/api/sensors/7/custom_sensors.html, which doesn't let you know that https://gazebosim.org/api/sensors/9/custom_sensors.html is the last version of the same page.

azeey, 2025-02-19:
I suppose we can add something in https://github.com/gazebosim/docs/blob/master/.github/workflows/nightly-upload.yml that injects the warning. It could read https://github.com/gazebosim/docs/blob/master/index.yaml to determine which API versions are outdated.

I've added the `help wanted` label.  Let me know if you'd be interested in taking this on.

EmmanuelMess, 2025-02-19:
> I suppose we can add something in [`master`/.github/workflows/nightly-upload.yml](https://github.com/gazebosim/docs/blob/master/.github/workflows/nightly-upload.yml?rgh-link-date=2025-02-19T21%3A22%3A26.000Z) that injects the warning. It could read [`master`/index.yaml](https://github.com/gazebosim/docs/blob/master/index.yaml?rgh-link-date=2025-02-19T21%3A22%3A26.000Z) to determine which API versions are outdated.
> 
> I've added the `help wanted` label. Let me know if you'd be interested in taking this on.

I don't have time right now.

shashank1300, 2025-04-13:
Hello @azeey @EmmanuelMess ,
I would like to work on this issue, if it's still possible. Can I get a bit more information please?