gazebosim/docs #573: Add related gazebo version to library APIs

EmmanuelMess, 2025-02-11:
Add related gazebo version to library APIs. For example, here is the sidebar for Gazebo sensors:
[image]

Tentative proposed change:

[image]

azeey, 2025-02-19:
Similar to https://github.com/gazebosim/docs/issues/572#issuecomment-2669789515 we could inject the collection(s) of a library in our build infrastructure. Note that a single library might belong to multiple collections. e.g. `gz-cmake3` was used by Garden and Harmonic.

SuperGops7, 2025-03-16:
Hi @EmmanuelMess @azeey can I take this issue up?

iche033, 2025-03-18:
sure please go ahead!

SuperGops7, 2025-03-20:
Hey @iche033  @azeey can you let me know where the api docs are and how we publish them?

iche033, 2025-03-20:
that section of the code is located [here](https://github.com/gazebosim/gz-cmake/blob/3509305eadb60b9fc78e91595d9adafa18d525fc/doc/doxygen/header.html#L37-L46) in gz-cmake. Every gz packages runs gz-cmake's [gz_create_docs](https://github.com/gazebosim/gz-cmake/blob/3509305eadb60b9fc78e91595d9adafa18d525fc/cmake/GzCreateDocs.cmake#L25) cmake function to create these doc pages, e.g. gz-sensors [gz_create_docs](https://github.com/gazebosim/gz-sensors/blob/810409d5fdcfc0cb0bf80f1188c6c7ba1db8eccc/CMakeLists.txt#L107-L110) call. 

To locally test your changes to gz-cmake, you can build another gz package (e.g. gz-sensors) in the same colcon workspace, and open`build/gz-common6/doxygen/html/index.html` to see the documentation page.

Create the PR, and once it's merged, the maintainers can help publish the docs.