gazebosim/docs #582: Rearange ROS 2 tutorials and ROS/Gazebo install

dvdmc, 2025-03-28:
This issue comes from here https://github.com/gazebosim/ros_gz/issues/708
We discussed reorganizing ROS 2 tutorials inside the Tutorials section. ROS/Gazebo installation would also be included in this tutorials. There are two problems to discuss:

- It seems that multi-level is not possible at the moment (see https://github.com/gazebosim/docs/blob/a574980d6aa853ac52af0a03f02f2e08d345fa89/build_multiversion.py#L127) so I added it alongside the Tutorials, which will also be more visible to ROS users. Should "ROS 2 tutorials" be on the top level or within Tutorials? The second would require implementing recursive children levels.

- ROS/Gazebo install is applied to all the docs, so it cannot be moved to the ROS 2 Tutorials as a subsection unless it's done for all the versions.

j-rivero, 2025-03-28:
> * It seems that multi-level is not possible at the moment so I added it alongside the Tutorials, which will also be more visible to ROS users. Should "ROS 2 tutorials" be on the top level or within Tutorials? The second would require implementing recursive children levels.

+1 for Going to "ROS 2 Tutorials" , don't need to mess with the docs page code at this point.

> ROS/Gazebo install is applied to all the docs, so it cannot be moved to the ROS 2 Tutorials as a subsection unless it's done for all the versions.

With "applied to all docs" do you mean that is in the level 1 of the index? If so, we have two options:
  1. Move that to be under the ROS 2 Tutorials
  1. Keep it the top level reference and add a new one under the ROS 2 Tutorials.

I would go with 2. if we can link the same document from two places.

dvdmc, 2025-03-29:
> With "applied to all docs" do you mean that is in the level 1 of the index? If so, we have two options:
> 
> 1. Move that to be under the ROS 2 Tutorials
> 2. Keep it the top level reference and add a new one under the ROS 2 Tutorials.
> 
> I would go with 2. if we can link the same document from two places.

The way Docs webpage currently builds is by having general pages to all Gazebo versions and adding version-specific pages (e.g., tutorials are specific). I can modify tutorials only for Harmonic since these documents are version specific. However, ROS/Gazebo install is in the group of general pages. Thus, there is no way of moving this file unless we add it to the version-specific documents for all versions. I can do it but since the idea was to modify only Harmonic I wanted to inform before.

kevshin2002, 2025-05-05:
Hi!

Is this still in the works? I am able to contribute to this as part of my early process of entering open source contributions.

Thanks!

j-rivero, 2025-05-05:
> Is this still in the works? I am able to contribute to this as part of my early process of entering open source contributions.

Thanks Kevin ! As far as I can tell the work is still pending. Feel free to start with it and we can merge efforts if @dvdmc has something started privately.

dvdmc, 2025-05-05:
Hi! @kevshin2002 I have been occupied lately with some deadlines but I was planning to get back to work. I didn't start anything so far, will try later today since you are interested.